apertus-open-source-cinema/axiom-firmware #12: Move scripts to /usr/local/script

jatha, 2017-11-03:
Currently, the apertus scripts are linked to `/usr/local/bin`with their file extension stripped off.
A much cleaner solution would be to link them with extension to a `/usr/local/script` directory and add this to the PATH.

sebix, 2017-11-03:
Why would that be cleaner?

jatha, 2017-11-03:
because it makes clear, that the tools are scripts. 
I personally dont mind that much and are open to other suggestions.
Maybe @herbertp wants to say something to this topic?

herbertp, 2017-11-03:
Because without extension (.sh, .py, etc) it is not clear that an interpreter is required.
And because scripts are not binaries (that's what /bin stands for) and thus are mostly architecture independent.

Best,
Herbert

sp-apertus, 2017-11-18:
This is taken care of in the latest version right? Close?

jatha, 2017-11-18:
according to https://github.com/apertus-open-source-cinema/beta-software/blob/388e12786b73034413258367aab2043547b5c4dd/build_tools/inside/update_inner.sh#L18, no