apertus-open-source-cinema/axiom-firmware #13: Add apertus ssh keys to the Camera

jatha, 2017-11-03:
Bertl suggested adding some default keys to the beta images. This would be convenient during development.

rroohhh, 2018-03-30:
A better solution would maybe be a overlay directory with files that get copied to the root filesystem during image creation, this would solve security / privacy issues connected with shipping a default key. It also allows users / developers to easily add their own system configuration to the images (default shell, shell config, custom bitstreams, own tools, etc).

sp-apertus, 2018-03-30:
Sounds like a good idea.

anuejn, 2019-03-05:
this is possible with the overlay feature introduced to the build system

closed 2019-03-05 by anuejn

rroohhh, 2019-03-11:
In addition to that see #65 for the remote debugging aspect of this.