apertus-open-source-cinema/axiom-firmware #20: install lighttp

sp-apertus, 2017-11-03:
it would be great if the AXIOM Beta would have a webserver by default in the image:
https://wiki.apertus.org/index.php/Installing_a_webserver_on_AXIOM_Beta

jatha, 2017-11-03:
Maybe this is related to #15, beacuse this needs lighthttpd installed as well (according to https://github.com/apertus-open-source-cinema/beta-software/tree/master/software/control_daemon)

BAndiT1983, 2017-11-12:
A question for the future: Can the CI system run a minifier on webGUI JS files? Also what about e.g. LESS processor for CSS?

davidak, 2017-11-15:
i would suggest to use nginx. it's easier to configure than apache2 (in the opinion of me and my collegues), high performance and widely used.

(i can help with configuration if needed)

sp-apertus, 2017-11-15:
lighttp was in particular selected because the Beta has limited resources and it worked very well so far.

jatha, 2017-11-17:
> A question for the future: Can the CI system run a minifier on webGUI JS files? Also what about e.g. LESS processor for CSS?

Yes, totally. This is completely doable, since the ci system can run arbitrary scripts. I would recommend add the minifier, you would like to use to the `build_tools/inner/dependencies.txt` and run it inside the `build_tools/inner/update.sh` :)

jatha, 2017-11-17:
@BAndiT1983  Should we do some reverse-proxying in the Webserver to expose the API of the control-daemon under the same port as `/api/*` or something similiar?

BAndiT1983, 2017-11-19:
Sorry for late response, was away from computer for some time. Daemon shouldn't be exposed at all to the outside, all the comms should be done through some interface like REST (see RESTServer) or similar. Only direct comm inteface daemon supports (at least at the moment) is flatbuffers and should only be used by modules available on camera (also see RESTServer for example).

jatha, 2017-11-19:
Is https://github.com/apertus-open-source-cinema/beta-software/pull/41/files#diff-f3d735f3b1c42274f606f43a4169e60cR26 ok?

BAndiT1983, 2017-11-19:
Looks good, just have to verify it later on my VM, as it's not really camera-dependent. WebGUI is still in progress, so the final details are still unknown but for now it's more than sufficient.

PR #41 (2017-11-19): install lighttpd and daemon in nightly images
closes #20 
closes #15
closed 2017-11-19 by sp-apertus