apertus-open-source-cinema/axiom-firmware #22: rename kick > start

sp-apertus, 2017-11-03:
to make it easier for people to understand what these scripts do rename:
kick.sh -> autostart.sh
kick_manual.sh -> start.sh
halt_manual.sh -> stop.sh

jatha, 2017-11-03:
start what? stop what? maybe start-$task and stop-$task would be more helpful?

sp-apertus, 2017-11-03:
kick_manual.sh currently initializes the image sensor communication, HDMI output generation and the continuous image capture for the video stream.

jatha, 2017-11-03:
then mabe all the scripts start_camera.sh stop_camera.sh and autostart_camera.sh?

sp-apertus, 2017-11-03:
since the service that runs these scripts is called "cmv12k" - and the scripts and service are image sensor specific we could also call it "start_cmv12000.sh"? But I understand this is very technical and people working with the camera might not care about the exact model name of the image sensor....

so "start_camera.sh" also sounds reasonable.... on the other hand the entire device is the camera and that is started before already....

jatha, 2017-11-03:
yeah you are right :). maybe start_cmv would be resonable as well. The term "cmv-" is included in many names and could already be known by users.  would vote for that.

anuejn, 2019-01-08:
done

closed 2019-01-08 by anuejn