apertus-open-source-cinema/axiom-firmware #23: prevent running start.sh twice

sp-apertus, 2017-11-03:
if you run the current kick_manual.sh (asked to be renamed to start.sh) twice it will crash the camera. To prevent this the idea was to create a temporary file in a tmpfs directory when the script is called. Also the script checks if said file exists and exits before doing anything else if it does. Then halt_manual.sh (asked to be renamed to stop.sh) should again remove the temporary file.

jatha, 2017-11-03:
isnt this a problem, that should be solved by the control daemon?

herbertp, 2017-11-03:
Yes, this is a problem to be addressed by the control daemon once it is up and running.
For now it would be good to touch a file in /run to avoid duplicate startup and interference between systemd autostart (cmv12k service) and manual script execution.

Best,
Herbert

davidak, 2017-11-20:
You can use [flock](https://linux.die.net/man/1/flock) to do that.

example:

```
#! /bin/bash

SCRIPT_BASE_DIR="/root/backup"

/usr/bin/flock -x -n $SCRIPT_BASE_DIR/scripts/backup_all_external.lock -c "$SCRIPT_BASE_DIR/scripts/backup_all_internal.sh"
```

anuejn, 2019-01-08:
@davidak fancy implementing this and trying it out in qemu?

feel free to ask any questions or if you need help to get started.

davidak, 2019-01-08:
@anuejn setting up qemu would take more time than implementing this.

Can you just create this file in the same directory as start.sh and test it out?

start_lock_wrapper.sh:
``` bash
#!/bin/bash
# create lock file to prevent multiple instance of the script running at the same time
flock -x -n /tmp/start.lock -c axiom-beta-firmware/software/scripts/start.sh
```

Change path of start.sh to it's actual path.

The lock file might fit better in /var/run ? Is this run by root?

When it works, i can create a PR.

anuejn, 2019-01-10:
It shouldnt be too hard to set it up. Just install docker, clone this repo and run `makefiles/docker-make.sh qemu-shell`. This will build the image with your changes and drop you into a shell of the camera in qemu.

If it doesnt work as described, please let me know. 

More image improvements will come for sure from your side, so it is nice, if you have a working setup anyways.

sp-apertus, 2021-11-30:
fixed: https://github.com/apertus-open-source-cinema/axiom-firmware/commit/79d44730833e98a1465741ada3fe29700026bbd7

closed 2021-11-30 by sp-apertus

davidak, 2021-11-30:
Why does a user have to run a shell script to have their camera working in the first place? Why is it not a systemd service that starts at boot?

Maybe that can be improved later.

sp-apertus, 2021-11-30:
We have that service but let users decide if they want to enable it or not.