apertus-open-source-cinema/axiom-firmware #78: add currently missing packages

rroohhh, 2019-03-05:
missing are  
- lsof
- mlocate
- stress
- screen
- minicom

gcolburn, 2019-04-07:
I would think these should probably be included since they are documented in the AXIOM Beta/Manual and are useful for wifi:

- iwlist
- wpa_supplicant

Used in the documentation here: [AXIOM Beta/Manual](https://wiki.apertus.org/index.php/AXIOM_Beta/Manual)

rroohhh, 2019-04-07:
sure, we should also consider making using wifi and creating wifi hotspots a bit easier for users.
I think `wicd` or `NetworkManager` could be good solutions for the client side.
For creating wifi hotspots we already have started some work in #85.

Also the documentation should probably be updated to for example use `ip` instead of `ifconfig` as the latter is deprecated.

commit db18a0a5 (2019-06-01): [arch] add some missing packages, fixes #78
closed 2019-06-01 by rroohhh