apertus-open-source-cinema/axiom-firmware #88: usb is broken

rroohhh, 2019-03-07:
`lsusb` output is empty and nothing happens when something is plugged in.

quanteksystemsinc, 2019-04-05:
I've observed the same behavior with `lsusb`. Any ideas what is required to fix it and an ETA? My router isn't in the greatest location to use ethernet frequently so getting wifi working would be a huge plus.

gcolburn, 2019-04-05:
I've observed the same behavior with `lsusb`. Any ideas what is required to fix it and an ETA? My router isn't in the greatest location to use ethernet frequently so getting wifi working would be a huge plus.

anuejn, 2019-04-05:
i think it is an issue with the devicetree and requires some experimenting. you are welcome to do so :).

gcolburn, 2019-04-05:
Good to know. If that's the case it should be an easy fix. I have another
SD card with working USB/Wifi from a few years back so I can look at that
devicetree.

rroohhh, 2019-04-05:
yeah only problem is, that we switched the kernel so the old devicetree is not 100% compatible with the new kernel

gcolburn, 2019-04-07:
USB is working with the device tree from the previous firmware I have for the beta which was using a 4.4 version of the kernel. The nightly build seems to be using 4.2. It would be great to get the changes reviewed/tested and into the latest build. Thanks!

Here is the diff:
```
diff --git a/boot/axiom-beta/devicetree.dts b/boot/axiom-beta/devicetree.dts
index 558e233..011778b 100755
--- a/boot/axiom-beta/devicetree.dts
+++ b/boot/axiom-beta/devicetree.dts
@@ -488,9 +488,11 @@
                        interrupt-parent = <&intc>;
                        interrupts = <0 21 4>;
                        reg = <0xe0002000 0x1000>;
-                       phy_type = "ulpi";
                        dr_mode = "host";
-                       usb-phy = <&usb_phy0>;
+                       phys = <&usb_phy0>;
+                       phy-names = "usb-phy";
+                       phy_type = "ulpi";
+                       vbus-supply = <&usb_vbus0>;
                };

                usb1: usb@e0003000 {
@@ -585,11 +587,19 @@
                };
        };

-    usb_phy0: phy@0 {
-               compatible = "ulpi-phy";
+    usb_phy0: phy0 {
+               compatible = "usb-nop-xceiv";
                #phy-cells = <0x0>;
                reg = <0xe0002000 0x1000>;
                view-port = <0x170>;
                drv-vbus;
     };
+
+    usb_vbus0: vbus0 {
+                compatible = "regulator-fixed";
+                regulator-name = "fixed-supply";
+                regulator-min-microvolt = <5000000>;
+                regulator-max-microvolt = <5000000>;
+    };
+
 };
```

rroohhh, 2019-04-07:
Awesome, thanks for looking into that, will test it and then report back.

The kernel should be version 4.20 not 4.2, can you confirm that?

rroohhh, 2019-04-07:
ok I looked into this a bit more.

There is a interesting patch in the xilinx forum about this:
https://forums.xilinx.com/t5/Embedded-Linux/Zynq-7000-USB-with-mainline-kernel-is-broken/td-p/744026

So setting the phy type to `usb-nop-xceiv` works, but I think it is not the proper solution as otg will stop working afaik.

We should try to incorporate the patch, and maybe adapt it so use `CI_HDRC_OVERRIDE_PHY_CONTROL` instead of what it is doing currently.

commit dbe36814 (2019-04-09): patch kernel to support zynq style vbus handling of usb, fixes #88
PR #99 (2019-04-13): patch kernel to support zynq style vbus handling of usb
- [x] tested on micro by @rroohhh 
- [x] teseed on beta by @anuejn 

closes #88
closed 2019-04-13 by anuejn