apertus-open-source-cinema/axiom-firmware #91: add documentation

rroohhh, 2019-03-28:
Add documentation about the firmware image
- basic login, accessing the camera over serial or ssh
- basic scripts and camera tools
- installing software and updating the image / software
- wifi / ethernet connection
- how to add your own software

Do we also need documentation for basic linux tasks like creating a user or the basic unix commands (`cd`, `ls`, `rm`, etc.)?
I think we should rely on external documentation for the basic linux stuff.

While writing this documentation the [manual](https://wiki.apertus.org/index.php/AXIOM_Beta/Manual) should be updated aswell.

rroohhh, 2019-03-28:
possible choice for the documentation of linux basics:
- http://wiki.lib.sun.ac.za/images/c/ca/TLCL-13.07.pdf
- http://linuxcommand.org/lc3_learning_the_shell.php

sp-apertus, 2019-03-30:
Started documentation for this specific firmware 2.0 for testers:
https://wiki.apertus.org/index.php/AXIOM_Beta_Firmware_Version_2.0