apertus-open-source-cinema/axiom-firmware #94: start.sh script lut_conf3 -> lut_conf

sp-apertus, 2019-04-04:
When I run sudo axiom-start.sh I noticed these errors:

./linear_conf.sh: line 10: ../processing_tools/lut_conf3/lut_conf3: No such file or directory
./linear_conf.sh: line 11: ../processing_tools/lut_conf3/lut_conf3: No such file or directory
./linear_conf.sh: line 12: ../processing_tools/lut_conf3/lut_conf3: No such file or directory
./linear_conf.sh: line 13: ../processing_tools/lut_conf3/lut_conf3: No such file or directory
./gamma_conf.sh: line 7: ../processing_tools/lut_conf3/lut_conf3: No such file or directory
./gamma_conf.sh: line 8: ../processing_tools/lut_conf3/lut_conf3: No such file or directory
./gamma_conf.sh: line 9: ../processing_tools/lut_conf3/lut_conf3: No such file or directory
./gamma_conf.sh: line 10: ../processing_tools/lut_conf3/lut_conf3: No such file or directory

I found the lut conf was actually ../processing_tools/lut_conf/lut_conf with no 3's.

sp-apertus, 2019-04-04:
Committed a fix: https://github.com/apertus-open-source-cinema/axiom-beta-firmware/commit/a0fb7d03f88961cc101f83703ee3a6ff4720bc71

Please check if this solves the issue.

rroohhh, 2019-05-29:
seems to be fixed

closed 2019-05-29 by rroohhh