apertus-open-source-cinema/axiom-firmware #98: `/etc/environment` is broken

rroohhh, 2019-04-12:
It does not work the way it is currently used. This manifests itself for example in `scp` not working.

It is unclear if there is something that actually does what we wanted `/etc/environment` to do.

anuejn, 2019-04-13:
well some sed on that file intead of that append should be fine :)

rroohhh, 2019-04-13:
Nope, because empty per default. I don't actually know where the default `PATH` is coming from.

anuejn, 2019-04-13:
ah shit... i see

rroohhh, 2019-05-24:
We should fix this sooner rather than later, as currently `scp` is borked and getting files on the camera is a pain.
 Just putting `PATH=/sbin:/bin:/usr/sbin:/usr/bin:/usr/axiom/bin` in `/etc/environment` works for me at the moment.
Its not that elegant, but maybe good enough?

anuejn, 2019-06-02:
seems to be okay-ish fixed for now

closed 2019-06-02 by anuejn