apertus-open-source-cinema/axiom-firmware #115: axiom-start.sh issue

sp-apertus, 2019-06-17:
during the axiom-start.sh routine this happens:

read buffer = 0x18390200
/usr/axiom/script/axiom-start.sh: line 52: ./rf_sel.py: No such file or directory
/usr/axiom/script/axiom-start.sh: line 53: ./pic_jtag_pcie.py: No such file or directory
Setting gain x1 (0,0x3eb)...

sp-apertus, 2019-06-17:
files are there in /opt/axiom-firmware/software/bringup-scripts/
but they are not executable (permissions) and no softlink in place from /usr/axiom/scipt/

rroohhh, 2019-06-18:
with which version (commit) is this happening?

sp-apertus, 2019-06-18:
I fixed it by creating 3 softlinks in the end.

Firmware version: "axiom-beta-b9cbb064.img.xz"

rroohhh, 2019-06-18:
Hmm strange that it broke while there were (seemingly) no relevant changes to `scripts` or `bringup-scripts`.

We should figure out some way to automatically test this is qemu.

sp-apertus, 2019-06-19:
We just realized there might have been a firmware update/flashing mix up with the two betas in the hub.

/etc/issue shows:

Build b9cbb064-dirty

sp-apertus, 2019-07-18:
any progress on this?

getting /usr/axiom/script/axiom-start.sh: line 53: ./pic_jtag_pcie.py: Permission denied now on remote beta where I had fixed it already but probably some work has been done in the meantime.

rroohhh, 2019-07-18:
I did no work on the remote beta, did anybody update / change the installed firmware?

sp-apertus, 2019-07-18:
I didn't but gsoc students use it regularly and rex just told me he flashed the firmware for Bertl with a new image around 3 weeks ago.

sp-apertus, 2019-09-16:
flashed the latest release and now have this issue again....

sp-apertus, 2019-09-16:
detailed info fix (as root):

cd /opt/axiom-firmware/software/bringup-scripts/
chmod +x pic_jtag_pcie.py
chmod +x rf_sel.py

ln -s pic_jtag_pcie.py ../scripts/pic_jtag_pcie.py
ln -s rf_sel.py ../scripts/rf_sel.py

sp-apertus, 2019-09-16:
hmm, now new error:

selecting RFW [bus A] ...
Traceback (most recent call last):
  File "./pic_jtag_pcie.py", line 10, in <module>
    from jtag import *
ModuleNotFoundError: No module named 'jtag'
Setting gain x1 (0,0x3eb)...

probably another missing softlink/file, what could the 3rd softlink I mentioned above I created could have been?

rroohhh, 2019-09-16:
seems like `jtag.py` is missing

sp-apertus, 2019-09-16:
yeah, that did it, thanks!

last command:
cd /opt/axiom-firmware/software/bringup-scripts/
ln -s ../scripts/jtag.py jtag.py

sp-apertus, 2020-01-31:
icsp.py needs the same treatment:

read buffer = 0x18390200
Traceback (most recent call last):
  File "./rf_sel.py", line 11, in <module>
    from icsp import *
ModuleNotFoundError: No module named 'icsp'
Traceback (most recent call last):
  File "./pic_jtag_pcie.py", line 28, in <module>
    jtag_tms(i2c, base, "01100")      # goto Shift-IR
  File "/opt/axiom-firmware/software/scripts/jtag.py", line 108, in jtag_tms
    jtag_seq(i2c, base, [0x02, 0x02], data, mod, val)
  File "/opt/axiom-firmware/software/scripts/jtag.py", line 70, in jtag_seq
    i2c.write_i2c_block_data(last + 1, mod, [val])
OSError: [Errno 6] No such device or address
Setting gain x1 (0,0x3eb)...
0x0

rroohhh, 2020-08-04:
Should be fixed atleast with fb4149086a4d8815bc7eb4d44a96a50c24b735c0

closed 2020-08-04 by rroohhh