apertus-open-source-cinema/axiom-firmware #120: make snap usable by operator?

sp-apertus, 2019-09-16:
<Bertl> with /dev/mem you have access to basically everything, so allowing that for non-root would basically make every user root

<Bertl> probably make the cmv_snap3 suid?

Sebastian tried the suid approach and it works

anuejn, 2019-10-13:
seems reasonable :)

anuejn, 2019-10-13:
some scripts need that too but scripts cant be setuided so we probably need a more generalized approach

sp-apertus, 2019-10-13:
I tried the suid approach and the execution worked but the resulting images where just appearing to be random data. I also tried the static binary of cmv_snap3 from the old firmware on the new firmware and that worked in the end.

sp-apertus, 2019-10-13:
Its well possible I made a mistake somewhere along the way so please test.