apertus-open-source-cinema/axiom-firmware #132: rcn-darkframe.py does not work with latest pypng

rroohhh, 2020-03-21:
It uses `read_pnm_header` from pypng, that seems to have been a helper for some CLI functionality that was moved into its own file. Version 0.0.18 of pypng is the last one to have it in pypng.

Maybe we just want to fix the version of pypng to that?

alex-ml, 2020-03-25:
Forcing the older version might interfere with future updates of other Python packages that use pypng, sooner or later.

The stackoverflow link from read_pnm also contains a version that only depends on numpy:
https://stackoverflow.com/a/31504580

Not tested, but if it works, I'd say it's a better approach.

Copying read_pnm_header from an older version of pypng should also do the trick. The one from pypng-0.0.14 appears to be reasonably short. Not tested either.

rroohhh, 2020-03-25:
> Forcing the older version might interfere with future updates of other Python packages that use pypng, sooner or later.

Yep, thats the reason this issue is not closed yet :)

Thanks for digging up some alternatives!

rroohhh, 2022-02-28:
properly fixed in e179890240c0f4dc7893290f625f2e4cef620b65

closed 2022-02-28 by rroohhh

sp-apertus, 2022-03-01:
Excellent that you found a proper solution!