apertus-open-source-cinema/axiom-firmware #161: Missing ssh keys for operator and root.

herbertp, 2020-08-16:
Both, 'operator' and 'root' user do not have ssh keys.

They should be generated on first boot and installed in the proper place (~/.ssh)

rroohhh, 2020-08-17:
Do you think using `ed25519` type keys would be ok? They are supposed to be more secure and the public key is a lot smaller than a `rsa` one. However not every ssh implementation supports them (`dropbear` being a notable one that doesn't support them, also any system using a openssh version from before 2014).

herbertp, 2020-08-17:
I think there is no problem in having elliptic curve keys on the beta but I wouldn't make that exclusive. As far as I know there is no problem having two or more different key types with openssh, so having one rsa, one dsa and one ed25519 should work just fine.

rroohhh, 2020-08-17:
Well I wouldn't do dsa :)
But having rsa and ed25519 seems fine.

herbertp, 2020-08-17:
Works for me.