apertus-open-source-cinema/axiom-firmware #162: Missing authorized_keys file for root

herbertp, 2020-08-16:
The 'root' user is missing an authorized_keys file with proper permissions.

We should also consider adding a key to allow remote access for testing and development (needs to be removed or the root ssh disabled on end-user installs).

rroohhh, 2020-08-17:
The `authorized_keys` file  was added in 490fee72438ded01e4438d772db72c8f642de973.

I think the easiest option for having some keys added to `authorized_keys` by default would be to just produce two images from the CI, one with the keys added and one without.
There probably are (or will be) other things aswell that we don't want / need on "user" images, but need on "development" images, so create two different images is probably necessary down the road anyways.

Which keys do you want to have added for the development image?

herbertp, 2020-08-17:
For now, the following key would work for me to get everything setup properly: [key.txt](https://github.com/apertus-open-source-cinema/axiom-firmware/files/5084021/key.txt)

We might want to have keys from all (main) developers at some point ...