apertus-open-source-cinema/axiom-firmware #165: Missing gateware hangs Beta on axiom_start.sh

herbertp, 2020-08-16:
In case the gateware for the Main SoC is missing (e.g. a broken symlink) or in the wrong format (.bit or .bin with wrong byteorder), FPGA manager cannot load the gateware sucessfully.

Unfortunately this is only commented by an output from 'echo' like this ...

`/usr/axiom/script/axiom_start.sh: line 9: echo: write error: No such file or directory`
`/usr/axiom/script/axiom_start.sh: line 9: echo: write error: Invalid argument`

... which then leads to a complete lockup once the memory mapped gateware registers are being accessed.

rroohhh, 2020-08-17:
We should definitely check the return code of the `echo` that load the bitstream and abort the `axiom_start.sh` script when that fails. 

Do you think that would be enough, or do you think translating the error message from `echo` into something clearer would also be needed?

herbertp, 2020-08-17:
I think aborting is probably enough (prevents the lockup :) but of course it would be nice to explain the possible reasons based on the error code.