apertus-open-source-cinema/axiom-firmware #179: Improve powerboard detection logic

rroohhh, 2020-12-02:
On a powerboard version 0.29 the powerboard version cannot be correctly determined when the mux (that only exists on version 0.29 and not 0.30) is held in reset.

So for the scripts to work, the pin controlling the reset has to be set unconditionally, atleast once, to check the powerboard version.

A quick fix is changing the `axiom_power_on.sh` script to always enable it:
```
#!/bin/bash
for id in 0x20 0x21; do
    i2c_test $id || continue

    i2c_set $id 0x14 0xFF
    i2c_set $id 0x15 0xFF
done

for id in 0x22 0x23; do
    i2c_test $id || continue

    i2c_set $id 0x14 0x4F
    i2c_set $id 0x15 0xBC
done
```

A more proper fix would be:
```
 00:13:55 <Bertl> probably the best way would be to take 22b7 high unconditionally (in gpio.py setup) and then check for the PB revision
 00:14:25 <Bertl> if we find no mux, we assume v0.30 and take it low again
```

imSanko, 2020-12-08:
I wanna work on to this **bug issue** can you assign it to me.

rroohhh, 2020-12-08:
@imSanko solving this issue needs extensive hardware testing we can only really do ourselves.