apertus-open-source-cinema/axiom-firmware #203: USB Serial Drivers/Kernel Module

sp-apertus, 2024-01-19:
Report notes from maxime:
http://irc.apertus.org/index.php?day=18&month=01&year=2024#6

https://bin.garbaye.fr/?b64702f42507a226#5oxkUw42MFB2g7rQzRJY77vMd5s3a39DtmW6b7QQDFaT

rroohhh, 2024-03-04:
the pastebin is expired, was there any relevant message in there?

sp-apertus, 2024-03-05:
lets ask maxime to share it again or remind us

emmahicks, 2024-03-14:
can't find what kind of error message I sent back then, but I can report that now we can see the ESP32 as /dev/ttyUSB0 ! (but it still didn't work with a microbit for example, but it's not that important ATM =)

emmahicks, 2024-03-14:
we are working here for the moment : https://git.erg.school/EA016/apertusIRLbxl and when we are happy with something we'll join this main git !