apertus-open-source-cinema/axiom-firmware #212: axiom_start: link training not working

Lynnaut, 2025-02-19:
Hi there,

i am currently experiencing issues while running the axiom_start script.
It seems that the link training does not work on my setup.
It will repeat in this loop forever as it never reaches a bit pattern of 0xFFFFFFFF.

Any help finding a solution for this would be appreciated.

Best regards

```
Feb 19 22:08:59 beta axiom_start.sh[11800]: Temp                   50.5697 °C
Feb 19 22:08:59 beta axiom_start.sh[11800]: Vccint                  0.9866 V
Feb 19 22:08:59 beta axiom_start.sh[11800]: Vccaux                  1.7871 V
Feb 19 22:08:59 beta axiom_start.sh[11800]: Vccbram                 0.9873 V
Feb 19 22:08:59 beta axiom_start.sh[11800]: Vccpint                 0.9866 V
Feb 19 22:09:00 beta axiom_start.sh[11800]: Vccpaux                 1.7871 V
Feb 19 22:09:00 beta axiom_start.sh[11800]: Vccoddr                 1.4897 V
Feb 19 22:09:00 beta axiom_start.sh[11800]: Vrefp                   0.0000 V
Feb 19 22:09:00 beta axiom_start.sh[11800]: Vrefn                   0.0000 V

Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a0] MXN_PG    I0+0^        [21a0] NE_PG     I0+0^        [22a0] IOE_PG    I0+0^        [23a0] IOW_PG    I0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a1] MXN_EN    O0+0^        [21a1] NE_EN     O0+0^        [22a1] IOE_EN    O0+0^        [23a1] IOW_EN    O0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a2] MCN_PG    I0+0^        [21a2] NN_PG     I0+0^        [22a2] RFE_PG    I0+0^        [23a2] RFW_PG    I0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a3] MCN_EN    O0+0^        [21a3] NN_EN     O0+0^        [22a3] RFE_EN    O0+0^        [23a3] RFW_EN    O0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a4] VIO_PG    I0+0^        [21a4] NW_PG     I0+0^        [22a4] B_#RST    O0+0^        [23a4] A_#RST    O0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a5] VIO_EN    O0+0^        [21a5] NW_EN     O0+0^        [22a5] u22#a5    I0+0^        [23a5] u23#a5    I1+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a6] MCS_PG    I0+0^        [21a6] WW_PG     I0+0^        [22a6] CSE_EN    O0+0^        [23a6] u23#a6    I1+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20a7] MCS_EN    O0+0^        [21a7] WW_EN     O0+0^        [22a7] CD'       I1+0^        [23a7] u23#a7    I1+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b0] MXS_PG    I0+0^        [21b0] SW_PG     I0+0^        [22b0] u22#b0    I1+0^        [23b0] u23#b0    I1+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b1] MXS_EN    O0+0^        [21b1] SW_EN     O0+0^        [22b1] u22#b1    I1+0^        [23b1] u23#b1    I1+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b2] CNW_EN    O0+0^        [21b2] SS_PG     I0+0^        [22b2] E_SPI_EN  O0+0^        [23b2] JTAG_EN   O0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b3] u20#b3    I1+0^        [21b3] SS_EN     O0+0^        [22b3] N_I2C_EN  O0+0^        [23b3] S_I2C_EN  O0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b4] u20#b4    I1+0^        [21b4] SE_PG     I0+0^        [22b4] N_SPI_EN  O0+0^        [23b4] S_SPI_EN  O0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b5] u20#b5    I1+0^        [21b5] SE_EN     O0+0^        [22b5] E_I2C_EN  O0+0^        [23b5] W_I2C_EN  O0+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b6] u20#b6    I1+0^        [21b6] EE_PG     I0+0^        [22b6] A_BUS_EN  O0+0^        [23b6] u23#b6    I1+0^
Feb 19 22:09:16 beta axiom_start.sh[11865]: [20b7] u20#b7    I1+0^        [21b7] EE_EN     O0+0^        [22b7] B_BUS_EN  O0+0^        [23b7] u23#b7    I1+0^
Feb 19 22:09:16 beta axiom_start.sh[11879]: ZED_5V                4.9219 V [1f80]         +17.8125 mV [1c8]   +890.62 mA
Feb 19 22:09:16 beta axiom_start.sh[11879]: BETA_5V               4.9414 V [1fa0]         +12.8125 mV [148]   +640.62 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: HDN                   3.1836 V [1460]          -0.0391 mV [fff]     -1.95 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: PCIE_N_V              2.8320 V [1220]          +0.3516 mV [009]    +17.58 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: HDS                   3.0859 V [13c0]          +0.0000 mV [000]     +0.00 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: PCIE_S_V              2.7930 V [11e0]          -0.0391 mV [fff]     -1.95 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: RFW_V                 2.8516 V [1240]          +0.1172 mV [003]     +5.86 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: IOW_V                 2.8711 V [1260]          +0.0000 mV [000]     +0.00 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: RFE_V                 2.9101 V [12a0]          +0.1562 mV [004]     +7.81 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: IOE_V                 2.9883 V [1320]          +0.0000 mV [000]     +0.00 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: VCCO_35               2.0508 V [ d20]          +0.3516 mV [009]    +17.58 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: VCCO_13               2.0703 V [ d40]          +1.0547 mV [01b]    +52.73 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: PCIE_IO               2.2461 V [ e60]          -0.0781 mV [ffe]     -3.91 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: VCCO_34               2.2461 V [ e60]          +1.0156 mV [01a]    +50.78 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: W_VW                  2.4023 V [ f60]          -0.0391 mV [fff]     -1.95 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: N_VW                  2.9101 V [12a0]          +0.0391 mV [001]     +1.95 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: N_VN                  1.6406 V [ a80]          +8.0078 mV [0cd]   +400.39 mA
Feb 19 22:09:17 beta axiom_start.sh[11879]: N_VE                  2.9101 V [12a0]          +2.7734 mV [047]   +138.67 mA
Feb 19 22:09:18 beta axiom_start.sh[11879]: E_VE                  3.2031 V [1480]          +0.0000 mV [000]     +0.00 mA
Feb 19 22:09:18 beta axiom_start.sh[11879]: S_VE                  1.9727 V [ ca0]          +0.0000 mV [000]     +0.00 mA
Feb 19 22:09:18 beta axiom_start.sh[11879]: S_VS                  2.9101 V [12a0]          -0.0391 mV [fff]     -1.95 mA
Feb 19 22:09:18 beta axiom_start.sh[11879]: S_VW                  1.9531 V [ c80]          -0.1562 mV [ffc]     -7.81 mA
Feb 19 22:09:20 beta axiom_start.sh[12225]: 0x00000254
Feb 19 22:09:20 beta axiom_start.sh[12226]: mapped 0x60000000+0x00400000 to 0x60000000.
Feb 19 22:09:20 beta axiom_start.sh[12226]: mapped 0xF8000000+0x00001000 to 0xB6FCC000.
Feb 19 22:09:23 beta axiom_start.sh[12226]: initial control adjustment ...
Feb 19 22:09:23 beta axiom_start.sh[12226]: adjusting out delay ...
Feb 19 22:09:23 beta axiom_start.sh[12226]: [00] = 15
Feb 19 22:09:23 beta axiom_start.sh[12226]: [01] = 16
Feb 19 22:09:23 beta axiom_start.sh[12226]: [02] = 15
Feb 19 22:09:23 beta axiom_start.sh[12226]: [03] = 16
Feb 19 22:09:23 beta axiom_start.sh[12226]: [04] = 16
Feb 19 22:09:23 beta axiom_start.sh[12226]: [05] = 16
[... 415 lines cut ...]
Feb 19 22:09:46 beta axiom_start.sh[12945]: [27] 0xFFFFFFFF center   => 16 (32,32)
Feb 19 22:09:46 beta axiom_start.sh[12945]: [28] 0xFFFFFFFF center   => 16 (32,32)
Feb 19 22:09:46 beta axiom_start.sh[12945]: [29] 0xFFFFFFFF center   => 16 (32,32)
Feb 19 22:09:46 beta axiom_start.sh[12945]: [30] 0x07FFFFFF msb = 26 => 10 (27,27)
Feb 19 22:09:46 beta axiom_start.sh[12945]: [31] 0x00FFFFFF msb = 23 => 07 (24,25)
Feb 19 22:09:46 beta axiom_start.sh[12945]: [32] 0x7FFFFFFF msb = 30 => 14 (31,31)
Feb 19 22:09:46 beta axiom_start.sh[12945]: checking bit pattern ...
Feb 19 22:09:46 beta axiom_start.sh[12945]: result = 0xFFFF6AFF
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20a0] MXN_PG    I0+0^        [21a0] NE_PG     I0+0^        [22a0] IOE_PG    I0+0^        [23a0] IOW_PG    I0+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20a1] MXN_EN    O0+0^        [21a1] NE_EN     O0+0^        [22a1] IOE_EN    O0+0^        [23a1] IOW_EN    O0+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20a2] MCN_PG    I0+0^        [21a2] NN_PG     I0+0^        [22a2] RFE_PG    I0+0^        [23a2] RFW_PG    I0+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20a3] MCN_EN    O0+0^        [21a3] NN_EN     O0+0^        [22a3] RFE_EN    O0+0^        [23a3] RFW_EN    O0+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20a4] VIO_PG    I0+0^        [21a4] NW_PG     I0+0^        [22a4] B_#RST    O0+0^        [23a4] A_#RST    O0+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20a5] VIO_EN    O0+0^        [21a5] NW_EN     O0+0^        [22a5] u22#a5    I0+0^        [23a5] u23#a5    I1+0^
[... cut ...]
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20b4] u20#b4    I1+0^        [21b4] SE_PG     I0+0^        [22b4] N_SPI_EN  O0+0^        [23b4] S_SPI_EN  O0+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20b5] u20#b5    I1+0^        [21b5] SE_EN     O0+0^        [22b5] E_I2C_EN  O0+0^        [23b5] W_I2C_EN  O0+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20b6] u20#b6    I1+0^        [21b6] EE_PG     I0+0^        [22b6] A_BUS_EN  O0+0^        [23b6] u23#b6    I1+0^
Feb 19 22:09:47 beta axiom_start.sh[12980]: [20b7] u20#b7    I1+0^        [21b7] EE_EN     O0+0^        [22b7] B_BUS_EN  O0+0^        [23b7] u23#b7    I1+0^
```

se6astian-pi, 2025-02-20:
Hi Lynnaut

The link training failing means there is a physical issue with the connection of the LVDS lanes between the image sensor and the FPGA/Microzed. This connection issue can be located in any of the high density connectors in the board stack. I will ask Herbert to chime in here and suggest the best way how to fix this.

Regards Sebastian

herbertp, 2025-02-20:
Hey Lynnaut,

Yes, as Sebastian already hinted, this is most likely a connection issue between the various PCBs.
The best way to trouble shoot this is to disconnect the PCBs starting from the front and give the connectors a thorough cleaning (check for flux residue and/or dust/dirt which might have sneaked in). Then reconnect the PCBs (they should snap in when the connection is formed) and rerun the training. Note the result values (result = 0xFFFF6AFF) after each clean and see whether it improved (got closer to the expected 0xFFFFFFFF).

Hope this helps,
Herbert

Lynnaut, 2025-02-20:
Thanks a lot for chiming in so quickly @se6astian-pi and @herbertp
After you both confirming the issue being a physical/electrical one i have disassembled the board stack to clean/re-seat everything.
Ive noticed after inspecting the connectors under the microscope quite a lot of flux residue between the legs that accumulated some dust/lint and most likely also moisture leading probably to some parasitic capacitance affecting the signal quality.
Sensor initialization now succeeds and i am getting a video signal.

closed 2025-02-20 by Lynnaut

se6astian-pi, 2025-02-20:
Perfect, well done :)